grblHAL/Plugins_motor #1: 5160 not configuring as expected

Cncninja, 2021-09-12:
I have a SKR1.4 board with TMC 5160 in SPI. when i poll the drives this is what is see below.  If I adjust the threshold  it seams to change my steps per mm, Olso the Vsense is set to .075 in the config but the poll shows N/A  the motor runs very choppy like is has no micro-stepping. I'm not sure what could be the issue. Im using IOsender 2.0.34, and the lpc176x_210809 build. 
[image]

terjeio, 2021-09-12:
>  If I adjust the threshold it seams to change my steps per mm

You see that change in the settings or in the speed or distance traveled - or both? Could it be due to the microstepping beeing set incorrectly?

> the Vsense is set to .075 in the config but the poll shows N/A

AFAIKT the 5160 does not have a vsense setting so it is ignored and reported back as N/A (Not Available).

> motor runs very choppy like is has no micro-stepping

Have you tried with a higher current setting?  Is the motor loaded (in a machine?) or not?

I'll have to wire up a SKR 1.3 board to check this again - I'll see if I can find time tomorrow.

Cncninja, 2021-09-13:
the settings do not change still shows the same steps per mm and 16 micro-steps, but the physical motion changes just like if I changed the steps per mm or micro stepping. 

the motor is just loose not connected to anything, I have 5160 in X and 2209 in standalone mode in Y for comparison. The 2209 is unbelievably smooth, the 5160 acts as tho it has no micro-stepping very choppy the SPI polls just fine and i can make changes but they just don't seem to change what it's supposed to. 

I have the bigtreetech 5160 V1.2 and all the Marlin videos and Bigtreetech pdf shows to set the 5160 Vsense as .075
So im not sure why.

Cncninja, 2021-09-13:
Sorry the Rsense value not Vsense

fairderben, 2022-02-08:
I am currently stuck experiencing the same Issue on my Trinamic2209 equipped SKR1.4 Turbo - M122 I makes the board unresponsive and forces a hard reset,
anything I could try?

terjeio, 2022-02-09:
@fairderben I have refactored the soft UART code and just committed the changes, hopefully it works now. I do not have a SKR 1.4 board to verify the code with, and there are [many different TM2209 drivers](https://3dwork.io/en/complete-guide-skr-v1-4-and-tmc2209/) out there requiring different links/wiring... I have tested with a SKR 1.3 board and Watterot drivers.

closed 2023-01-29 by terjeio