grblHAL/Plugins_motor #2: fatal error: trinamic2130.h

drom89, 2021-10-03:
So when compiling thru MCUX i get this error:

/src/trinamic/TMC2130_I2C_map.h:44:10: fatal error: trinamic2130.h: No such file or directory
   44 | #include "trinamic2130.h"
      |          ^~~~~~~~~~~~~~~~

So i just changed 44 | #include "trinamic2130.h" to #include "tmc2130.h" and its working, but im not sure if its correct.

drom89, 2021-10-03:
Anyway i have BTT TMC2209 with BTT SKR 1.4T and with #define TRINAMIC_ENABLE 2130
it still dont communicate with drivers :(
I think 2209 dont have SPI and only have UART communication.

terjeio, 2021-10-03:
> /src/trinamic/TMC2130_I2C_map.h

This file (and the corresponding .c file) should no be there, delete them. I have removed them in the repo already.

> I think 2209 dont have SPI and only have UART communication.

Correct, and the LPC176x driver does not support TMC2209 as BTT connected the UART lines to pins that cannot be routed to hardware UARTs. To support them software UARTs has to be implemented, and I have no plans to do so myself. I'll accept a pull request if you do so, ref. [this comment](https://github.com/grblHAL/LPC176x/issues/5#issuecomment-932989702).

drom89, 2021-10-03:
Thanks for reply, my programing skills are really bad, so i will order 2130 and 5160 and use them with SPI

terjeio, 2022-06-15:
FYI soft UART support was added to the LPC17xx driver in january.

closed 2022-06-15 by terjeio