grblHAL/Plugins_motor #4: TMC Config does not seam to match register

Cncninja, 2021-10-31:
I have a skr1.4 turbo LPC1769 board running the latest build as of 10/31/21.  I have TMC-2160 drivers installed. and jumpers in spi mode. IOsender auto recognizes them and added the Trimatic tab and I can see the driver status so everything seems to be configured corectly.  in the LPC Grblhal build TMC5160 is defined. 

The issue I have is the steppers are loud and choppy/vibrate. completely different from when I use them with other firmware. it appears that the Rsence and some other configurations are not being applied correctly.  based on the TMC status screen also if i jog and re-poll the tmc's the settings seem to change I have attached the Grblhal trimatic build config and the TMC screen from IOsender.  I don't see a few of the lines populated and the values seem off from what I see online for 5160's I think it has to do with the rsense value not showing. 
[image]
[image]
[image]
Thank you for your help

terjeio, 2021-10-31:
TMC2160 and TMC5160 has the same register structure?

R_SENSE is the same as Marlin(?), 75 milliohm = 0.075 ohm.

Cncninja, 2021-10-31:
Humm, then I'm not sure what's wrong, do the values look correct?

terjeio, 2021-10-31:
Assuming the 5160 and 2160 register layouts are the same:

> do the values look correct?

I believe so.

Have you tried to set the hybrid threshold with M913? It can also be compiled in by changing PWM_THRESHOLD_VELOCITY in [trinamic.h](https://github.com/grblHAL/Plugins_motor/blob/master/trinamic.h). Other parameters can be changed there as well.

Setting a higher motor currents helps? This is done by $-settings per axis.

Note that most of the default values in _tmc5160.h_ is overridden by settings in _trinamic.h_ and $-settings. The settings in _tmc5160.h_ is mainly aimed at those who want to use the low level API directly.