grblHAL/Plugins_motor #7: TMC_DEV compiling error

rcKeith, 2022-12-12:
Hi
I'm having issues with the SKR Pico Board where the steppers run very slowly. I'm using 1/8th micro-stepping with a 1.5mm lead screw pitch and 1066 steps/mm.  This works fine on my foam cutter using Grbl-Mega-5x using an Arduino Mega 2560. I've been running this for 3 years and built many RC model aircraft.  I decided it was time to look at 32-bit with the RP2040.
I just have the motor on the bench at the moment and even if I try different step/mm settings and micro-stepping they turn very slow, so slow I didn't think they were turning at first.  I've tried my G-code sender and ioSender and both show the same slow running in the DRO's.
I decided to enable TMC_DEV to further investigate but I get compiling errors. The first error being
**********************************************
_[build] C:\Users\khowl\Documents\Code\RP2040\motors\trinamic.c:776:111: error: 'stepper[report.sg_status_motor]' is a pointer; did you mean to use '->'?
[build]   776 |     sprintf(sbuf, "[SGPARAMS:%ld:%d:%d:%d]" ASCII_EOL, report.sg_status_motor, stepper[report.sg_status_motor].coolconf.reg.sfilt, stepper[report.sg_status_motor].coolconf.reg.semin, stepper[report.sg_status_motor].coolconf.reg.semax);
*******************************************************
[TMC_DEV.txt](https://github.com/grblHAL/Plugins_motor/files/10205575/TMC_DEV.txt)

I've attached the compiler output which has more errors.  Without TMC_DEV enabled it compiles without any errors.

As an additional test, I have one of the little Pico Boards and if I use PicoBoB as the board in my_machine.h the DRO's show the normal speed in the DRO's.  So I suspect it's something in the Trimanic 2209 drivers causing the issue.

Any help greatly appreciated.
Keith

terjeio, 2022-12-12:
The TRINAMIC_DEV option should be removed? After adding TMC2209 support it is no longer easy to add generic support for debugging as ther register structures differs significantly from the SPI drivers.
For now use the [M122 output](https://github.com/grblHAL/Plugins_motor#m122---output-debug-info-or-reset-driver) for debugging. Is this displaying sensible values?

rcKeith, 2022-12-12:
Hi 
Many thanks for responding so quickly. The output from M122 is below.  I'm not that familiar with the Trimanic 2209 so does this look OK?

M122
[TRINAMIC]
                      X       Y       Z       U
Driver          TMC2209 TMC2209 TMC2209 TMC2209
Set current         500     500     500     500
RMS current         489     489     489     489
Peak current        691     691     691     691
Run current       15/31   15/31   15/31   15/31
Hold current       7/31    7/31    7/31    7/31
CS actual          7/31    7/31    7/31    7/31
PWM scale           233     255     255      37
vsense          1=0.180 1=0.180 1=0.180 1=0.180
stealthChop        true    true    true    true
msteps                8       8       8       8
tstep           1048575 1048575 1048575 1048575
pwm
threshold             0       0       0       0
[mm/s]                0       0       0       0
OT prewarn        false   false   false   false
OT prewarn has
been triggered    false   false   false   false
blank time            1       1       1       1
hysteresis
-end                 -1      -1      -1      -1
-start                1       1       1       1
Stallguard thrs       0       0       0       0
DRIVER STATUS:
stallguard     
sg_result             6      22      26       0
fsactive                                       
stst                  *       *       *       *
s2gb                                           
s2ga                                           
STATUS REGISTERS:
 X = 0xC0:07:00:00
 Y = 0xC0:07:00:00
 Z = 0xC0:07:00:00
 U = 0xC0:07:00:00

terjeio, 2022-12-13:
The M122 output looks good, what is your $4 setting value?
If you change the microstepping is that having any effect on the speed?

rcKeith, 2022-12-13:
Hi
Unfortunately, the SKR Pico board died today. I suspect this may have been a faulty board. I have returned it to amazon and will get another hopefully this was the issue.  I'll update you again as soon as the new board arrives.
Many thanks
Keith

terjeio, 2022-12-22:
> During some testing, I found it took 30 secs to move 1.5 mm with a microstep of 1/8th and a steps/mm of 1066 for 1.5 mm pitch lead screw.

With which feed rate?

> As you may be aware the TMC2209 on the SKR Pico are embedded so there are no jumpers to set the micro-stepping. I wondered if this was the problem.

No, microstepping (and many other parameters) are set by the firmware. What could be the problem is that the driver do not output the correct number of pulses?

FYI I do not have a SKR Pico board available, and for the time beeing I do not have access to my TMC test platform either.

rcKeith, 2022-12-22:
Hi
Thanks for the reply.
I was testing at 100mm/min feed rate with G1 X 1.5 F100 which did turn the stepper motor 1 complete revolution but very slowly.
I was thinking maybe as you have said it could be the pulses.  Could you point me to the area of the code where this is calculated and I will try and debug.
I did manage to get a Pi Pico set up as a probe and run OpenOCD but I think it was looping waiting for me to connect. I was trying this on the same laptop with both connected but maybe I will have better luck running the probe on one laptop and have the SKR connected to another with ioSender.
Many thanks
Keith

terjeio, 2022-12-22:
The main stepper interrupt is generated by a [PIO program](https://github.com/grblHAL/RP2040/blob/6ef6c4d2891c56c72a081d87467149a5972e80ea/driverPIO.pio#L23-L68), search for `stepper_timer_sm` in _driver.c_ to find where it is loaded and timing is set .

rcKeith, 2022-12-28:
Hi
I've come to the conclusion its something to do with the feedrate.  To confirm my board was OK  I  installed  Klipper on a Raspberry Pi 4 and connected the SKR Pico to it.  After some configuration, I had the stepper motors working as expected.  So with G1 X1.5 F100 the motor did that in a couple of seconds.  With 1/16 micostepping and a 1.5 mm pitch lead screw.  I ran the DUMP_TMC STEPPER=stepper_x and got the output in the file attached.

[DUMP_TMC STEPPER.txt](https://github.com/grblHAL/Plugins_motor/files/10315793/DUMP_TMC.STEPPER.txt)
I then compared the setting and most were the same and I changed some I've attached the tmc2209.h with the changes and the klipper setting in the comments
[tmc2209.h.txt](https://github.com/grblHAL/Plugins_motor/files/10315875/tmc2209.h.txt)

When watching the realtime output from ioSender the FS field shows 100 if I send G1 X1.5 F100 but the motor turns very slowly taking about 15-20 seconds to complete the command.  The microstepping is correct because the motor turns exactly 1 turn for X1.5 as my leadscrew is 1.5mm pitch.  If I give G1 X1.5 F500 it moves at a similar speed to Klipper using G1 X1.5 F100

It seems there is some problem with the feed rate somewhere but my coding skills are not enough to identify where it is.  I've spent many hours examing the code but can't seem to find the problem.

Hope you had a good Christmas and may thanks for your help
Keith

terjeio, 2022-12-29:
Can you try with changing `pio1` to `pio0` in [this line](https://github.com/grblHAL/RP2040/blob/74b2602040e1da0ee861ccd637c9ab4cc18cf7ea/driver.c#L770)?

rcKeith, 2022-12-29:
Hi
Yes, that did it, thank you so much.  If you have time can you briefly explain how that fixed it? Just curious if you are too busy doesn't matter.
Thanks
Keith

terjeio, 2022-12-29:
Pio1/sm0 is for the main stepper timer program, the M3 motor (A-axis)  wrote data to that when it should not (it should write data to Pio0/sm0. sm0 = state machine 0.

rcKeith, 2022-12-29:
Hi
Thanks I'll get the board running on my machine and let you know. 
Happy New year
Keith

rcKeith, 2023-01-07:
Hi 
Have another little problem with the 4th axis. XYZ all work perfectly. But the 4th Axis 'A' doesn't move. If I send a jog command sometimes you can see a very tiny move as if the motor is being enabled.  I've checked through the pin map against the SKR Pico schematic  [https://github.com/bigtreetech/SKR-Pico/blob/master/Hardware/BTT%20SKR%20Pico%20V1.0-SCH.pdf](url)
and it all checks out OK.  I wonder if there is still an issue with the state machine.  I've looked through the code to the best of my ability but nothing stands out.
All the limit pins trigger fine including A but maybe that's not relevant 
Many thanks.
Keith

terjeio, 2023-01-08:
What is your $4 setting? It should be 15.

rcKeith, 2023-01-08:
Hi
$4 is set to 15.  Just to confirm the 4th axis was OK I flashed Klipper on to the board and tested that the 4th axis moved OK.  Which it did so I suspect there is a problem somewhere.  I've attached the settings from ioSender's backup.
[settings.txt](https://github.com/grblHAL/Plugins_motor/files/10369375/settings.txt)
Thanks for your help
Keith

terjeio, 2023-01-09:
Are able to check signals?
First the stepper enable, does it go low when moving axes? You may also check if the motor is active by trying to turn it.
If it does can you then check the step signal with a scope?

I cannot check these myself before I am back home later in January.

rcKeith, 2023-01-09:
Hi
I managed to do some signal testing just by connecting a led to pins 11 for X and pin 14 for A.  It showed a very faint light when I jogged X but nothing when I jogged A.  So I believe pin 14 for A is receiving any step signal.
Not the best way to do this but I think it shows the problem.
Hope that this is some help to the problem,
Thanks
Keith

terjeio, 2023-01-10:
I cannot spot any obvious bugs so this have to wait until I am back home.
Until then you may try with not remapping A-axis to U to see if that makes any difference?

rcKeith, 2023-01-10:
Hi
I tried not remapping and the results were the same. Thanks for checking. 
Keith

terjeio, 2023-01-13:
> it seemed to be missing the the forth axis with #if N_ABC_MOTORS > 1

Good catch, this line should be `#if N_ABC_MOTORS` only.

> But I do get a warning during the compilation

This is likely you modifying the code by accident, the line is

[static axes_signals_t next_step_outbits;](https://github.com/grblHAL/RP2040/blob/e66262ce5d6898174bb558987d3826a7a08469a5/driver.c#L155)

However, the variable is not referenced so should be removed - a leftover from the early days of the driver?