grblHAL/Plugins_motor #8: Issues with 2209s

troth530, 2023-01-23:
Hi, I'm having 2 main issues.  1) Enabling hybrid mode for switching between stealthchop and spreadcycle and 2) Stallguard for sensorless homing.

Hybrid Issue:
I cannot determine if TMC2209_TPWM_THRS in tmc2209.h  is being honored or overwritten by PWM_THRESHOLD_VELOCITY in trinamic.h.  TPWMTHRS from the 2209 datasheet is the TSTEP at the desired velocity, I cannot determine if you account for the unit conversion between velocity and TSTEP properly.  Also after setting PWM_THRESHOLD_VELOCITY a broken piece of code is exposed here: https://github.com/grblHAL/Plugins_motor/blob/d429c50ece14e36f3ca675ebca8f6c9b15c2e7f2/trinamic.c#L670

The error is 
../main/motors/trinamic.c:670:98: error: 'motor_settings_t' {aka 'struct <anonymous>'} has no member named 'steps_per_mm'

Stallguard Issue:
When enabled, stallguard seems to be oversensitive even during a typical jog.  There is persistent stalling after issuing M122 x S1.  M122 x S1 shows SG_RESULT ranging from 0 to 300 which almost implies either it's configured wrong or stallguard cannot be used except during homing. Using the calculator below, TMC2209_TPWM_THRS  is set to 38 (corresponding to 6000mm/min).  I've tried TMC2209_COOLSTEP_THRS ranging from 100 to 200.  The datasheet says diag is only pulsed when TCOOLTHRS ≥ TSTEP > TPWMTHRS.  I used this calculator for calculating Txxx.  I've been trying to unravel this mystery, but I need your help in figuring it out.  I know for instance your default settings do not satisfy the conditions gating diag being pulsed and perhaps the unit consistency is a problem as everything must be implemented in units of TSTEP.  

[TMC2209_Calculations.zip](https://github.com/grblHAL/Plugins_motor/files/10476531/TMC2209_Calculations.zip)

Thank you!

wakass, 2023-01-23:
This could be related to the pwm_autoscale being disabled on enableStallguard. I created an issue on this: https://github.com/terjeio/Trinamic-library/issues/6 
Would be good to get some confirmation.

troth530, 2023-01-24:
@wakass, that helps part of the issue (I believe).  Thank you!

Still not sure about units, I also notice there are raw values being read, but not sure what is being written.  I still believe unit consistency should be required for trinamic internal algorithms at least for this condition of this comparison  tcoolthrs < tstep < tpwmthres.

wakass, 2023-01-24:
@troth530 Regarding your first question, first keep in mind that tpwm threshold is the threshold where pwm-mode/stealthchop is operational (TSTEP ≥ TPWMTHRS), i.e. velocity_step < velocity_tpwmthrs. Above the velocity_tpwmthrs, spreadcycle is enabled (and pwm regulation disabled). In that regards, a 'desired' velocity as you mention 

>TPWMTHRS from the 2209 datasheet is the TSTEP at the desired velocity

is correct, but the desired velocity will likely not be the velocity at which you will be driving. It needs to be set a-priori to where you expect the pwm regulation of stealthchop to not be reliable. Anyway, i'm being a little pedantic here.

The trinamic plugin sets the tpwm_threshold also in the code below, which I think gets run after initialization. But i'm not too sure about that, you can put some debug output lines to make sure. In that sense I think the code that is giving you the error can be removed entirely. Do note that the same threshold velocity is set for all axes.

 https://github.com/grblHAL/Plugins_motor/blob/d429c50ece14e36f3ca675ebca8f6c9b15c2e7f2/trinamic.c#L554

Conversion from speed to time step is achieved here:
https://github.com/terjeio/Trinamic-library/blob/9eae92beab071f00d188f9da25672ab9f2aaa116/tmc2209.c#L190

M122 also gives debug output on tpwm btw.

troth530, 2023-01-26:
@wakass can you provide more details about M122 debug output, what is the syntax?
So far I have confirmed pmw_autoscale fixes the stallguard issues- great find.  I still need to confirm hybrid behavior. 
Thank you!

wakass, 2023-01-26:
See this issue here for output from M122, https://github.com/grblHAL/RP2040/issues/41#issuecomment-1379449288

Note the pwm section with the threshold and mm/s setting. This should give you an idea.

Syntax for M122 is described in this very repository: https://github.com/grblHAL/Plugins_motor

troth530, 2023-05-08:
Sensorless homing is fixed with re-enabling pwm_autoscale.  
Hybrid mode is fixed  adding float steps_per_mm; to struct in motor_settings_t in motors/trinamic.h

closed 2023-05-08 by troth530