grblHAL/Plugins_motor #9: Setting wrong motor stallguard settings on sensorless per-axis homing under some conditions.

wakass, 2023-01-23:
Found some odd behaviour when trying per axis homing, while having sensorless enabled on partial axes. I suspect the trinamic_on_homing function has some strange logic leading to this. https://github.com/grblHAL/Plugins_motor/blob/d429c50ece14e36f3ca675ebca8f6c9b15c2e7f2/trinamic.c#L1184

To trigger the bug it is required to enable the tmc drivers on all drivers (3 in my case). 

The bug seems triggered when setting partial sensorless homing axes using $339:
e.g. (debug added in my build)

This is the correct output when enabling all sensorless axes:

$339=7
ok
$HX
[SG: Motor: 2]
[SG: TCOOLTHRS set: 135]
[SG: Sens set: 80]
[SG: Motor: 1]
[SG: TCOOLTHRS set: 281]
[SG: Sens set: 30]
[SG: Motor: 0]
[SG: TCOOLTHRS set: 281]
[SG: Sens set: 100]
 

Its sets only the last motor parameters (motor 2) and stops.
$339=4
ok
$HX
[SG: Motor: 2]
[SG: TCOOLTHRS set: 135]
[SG: Sens set: 80]

troth530, 2023-01-27:
@wakass can you please identify how you added the extra debug output.

wakass, 2023-01-27:
I just peppered the trinamic and tmc drivers with hal.write.stream.  
[trinamic_debug.patch](https://github.com/grblHAL/Plugins_motor/files/10521588/trinamic_debug.patch)
[motors_debug.patch](https://github.com/grblHAL/Plugins_motor/files/10521589/motors_debug.patch)

terjeio, 2023-01-27:
There is a bug, only the axis/axes homed in the active cycle should have the parameters changed.
`$HX` with `$339=4` (Z axis) should not modify any.

https://github.com/grblHAL/Plugins_motor/blob/de41f8aa50ba8c5b22a0141a7e0b1bd535f4774f/trinamic.c#L1292

should be:

`    axes.mask &= (driver_enabled.mask & trinamic.homing_enable.mask);`