grblHAL/Plugins_motor #13: Verification of change for TRINAMIC_R_SENSE value?

bdbell, 2024-03-31:
Is there a good way to check to see if the setting for TRINAMIC_R_SENSE is changed from the default 0f 75 mOhm after the firmware is built?  I am using the BTT 5160 Plus external driver modules and they use a 22 mOhm current sense resistor rather than the standard 75 mOhm.  I have added `-D TRINAMIC_R_SENSE=22` to platformio.ini as I am using platformIO to perform the builds but with testing I am still drawing over 3.5A for a driver when the current is set to 1A.  Is there a way to verify that the value is actually included properly?

terjeio, 2024-04-01:
Try with different r sense values and check the if _Run current_ changes in the `M122` report?

bdbell, 2024-04-04:
I am unclear on what `Run Current` in the M122 report is supposed to indicate.  It appears to be a ratio but I have found not documentation on what it is supposed to represent.  the `Hold Current` makes a little more sense as it is a percentage of the run current.  Is there somewhere I can reference to understand those values?

terjeio, 2024-04-04:
There are [links to the datasheets in the readme](https://github.com/grblHAL/Plugins_motor#339---sensorless-homing-enable).