grblHAL/Plugins_motor #18: TMC 2209, abnormal behavior in "grblHAL build/patch 20250103"

raoullector, 2025-01-06:
Hi,

Homing no longer works on a ganged axis, the holding torque is almost nonexistent, and the $180-$183 and $190-$193 parameters have disappeared.

I updated today (2024/01/06) the GrblHal firmware, from "build 20241127" to "grblHAL build/patch 20250103" compiled with Webbuilder.
Of course, I took note of the warning relative to the settings reset and right after the update, I restored the settings of my CNC.
After a few tests, I had to downgrade the firmware to "build 20241127" in order to recover a proper functioning of the machine.

Here is what I observed with the build "grblHAL build/patch 20250103" :
- The ganged axis (X in my config) refused to perform the homing, not moving and made the same noise as when losing steps.
- Even when setting motor current at standstill at 100% ($21x), the M122 report displayed 15/31 for the hold current. Worse, after three seconds, the loss of torque on all axis was complete. Of course, I want my steppers not to be disabled when stopped ($37=15).
- The $180-$183 and $190-$193 parameters have disappeared, which may not be an issue since I was never able to get anything useful from them (I use 24 and 25 instead).

The controller of this machine is a SKR 3 EZ (STM32 H723) which drives 5 TMC 2209, two of them being ganged.

Regards

raoullector, 2025-01-07:
Hi Terje,

Here is the current config, i.e., the one I use for "build 20241127".
Just remove the $180-$183 and $190-$193 parameters to get the exact conf I applied with "grblHAL build/patch 20250103".
[Paramétres_GRBL_V04.txt](https://github.com/user-attachments/files/18330269/Parametres_GRBL_V04.txt)

For reference, I provide you with the firmware "grblHAL build/patch 20250103" the WebBuilder generated yesterday.
[firmware_SKR3_H723_bootloader_241222.zip](https://github.com/user-attachments/files/18330324/firmware_SKR3_H723_bootloader_241222.zip)

Also, you asked me whether both motors failed : unfortunately, I can’t tell you because the X ganged axis has its motors running in opposite directions. The only thing I am sure of is that the axis didn't move and made harsh noise.

I don't think that the disabling of the current pot may have something to do with the loss of torque : the TMC2209 drivers I use are fully configurated through UART. Physically, I even believe they come without a potentiometer (EZ2209 from BTT).

Regards,

terjeio, 2025-01-07:
I am still not able to replicate using the STM32F4xx driver and regular Silent Stepsticks. I do not have a BTT SKR3 board so cannot test with that, perhaps @dresco can?
M122 output after homing fails could be helpful, also try with less current in case overtemp is triggered.

dresco, 2025-01-07:
Yep, I can put the 2209 modules in my SKR3 EZ for testing.

Is just on the bench for development testing, but has 3 steppers attached. Will look for differences in M122 and apparent holding torque.

Am a bit confused by the version numbers above though, latest released H7 build was driver version 241221, and core version 20241222 (or are submodules getting bumped outside of the H7 repo for the webbuilder?).

terjeio, 2025-01-07:
> (or are submodules getting bumped outside of the H7 repo for the webbuilder?).

No, perhaps I should put a note in the Web Builder about the H7 following a different update schedule?

Perhaps [this commit](https://github.com/terjeio/Trinamic-library/commit/4b86246ef12dcf5d92906e3f077bc433d9e8a8cb) is the reason? When I test with Stepsticks at max current (with heatsinks) they run too hot and overtemp is triggered.

raoullector, 2025-01-07:
Great, I’m glad the issue has been identified.

@dresco , no thanks, I’ll pass and don’t have any urgent need. In fact, I was just taking advantage of a free moment to update the firmware; I will try that again later…

However, I’d be happy to get the info when the WebBuilder is operational again for STM32H7xx configurations.