grblHAL/Plugins_motor #19: Motors not powered, driver issue?

raoullector, 2025-01-18:
Hi,

I’m not sure if this is the right section, but I feel like my issue might not be related to the drivers themselves. I have a 4-axis CNC machine with one coupled axis, controlled by an SKR 3 (STM32H723) using EZ2209 drivers. I’m running GrblHAL, using either versions generated with the WebBuilder or compiled myself (STM32CubeIDE). However, I decided to replace my five EZ2209 drivers with EZ5160 Pro drivers. From that point on, nothing works anymore :

1) I install the new EZ5160 drivers, recompile the firmware to work with them, and power on the machine. Result: none of the steppers are powered, and M122 returns "M122 => MSG: Warning: Could not communicate with stepper driver!"

2) I try all the $RST versions. Result: it still doesn’t work.

3) I remove the drivers one by one in case one of them is defective. Result: still out of work.

4) I decide to revert to the previous working configuration, reinstall the EZ2209 drivers, and reload the old firmware. Result: fxxk! It works no more! The only small positive point is that the motors are slighty powered, but I can't change the current nor can I  communicate with the drivers. Just like with the EZ5160, M122 returns "M122 => MSG: Warning: Could not communicate with stepper driver!"

I feel like the different firmware updates have conflicted with each other and are causing interference. For example, I noticed that the value of the BUILD_INFO symbol, which is supposed to display with $I, did not always match what I had specified during compilation.

What do you think about this, considering that what I want to achieve is getting the EZ5160 to work?

terjeio, 2025-01-18:
Note that @dresco has to merge changes before they become available for the STM32H7xx driver.

dresco, 2025-01-19:
> Note that [@dresco](https://github.com/dresco) has to merge changes before they become available for the STM32H7xx driver.

Thanks, will try to update the H7 repo tomorrow.. 

@raoullector Would be useful to investigate the initial failure. Do you have any means of observing the SPI communication - logic analyser or scope? (The signals are easy to reach on the SKR3 via the 0.1" headers around the EZ sockets).

raoullector, 2025-01-19:
Hi Jon,

No, I don't have any tools to analyze the communication between the board and the drivers.

However, I plan to modify the code to trace which drivers are waiting for initialization. I want to ensure that it is not always the same driver(s), in which case the problem would probably be related to my setup.
I'll keep you informed of the results.

dresco, 2025-01-20:
Yep, could be something environmental.

As an example, I moved to these [integrated closed loop steppers](https://www.omc-stepperonline.com/icl-series-integrated-motor) last year. Overall have been impressed with them, but over the last couple of months they would intermittently power up with the fault output enabled (always rectified by a power cycle or two). I eventually realised that it appears to be temperature dependant, below about 5degC in the workshop & the trouble starts..

terjeio, 2025-01-20:
> However, I plan to modify the code to trace which drivers are waiting for initialization

If  `$338` is available you can enable one driver at a time to find out which fails:

[image]

FYI I have just committed [another update](https://github.com/grblHAL/core/blob/master/changelog.md#202150120) that adds a command for outputting driver status, I have not tested it much yet and its underlying functionality for the Trinamic drivers can be improved. For another time...

dresco, 2025-01-20:
fyi, have just updated the H7 driver with latest F7 changes + submodules.
Have **not** attempted to add the per-axis motor fault or reporting code (from the F4 driver) yet though..

terjeio, 2025-01-20:
Web Builder now updated to latest H7 code. 

> Have not attempted to add the per-axis motor fault or reporting code (from the F4 driver) yet though..

Changes to the driver itself is not needed for reporting `$SDS` status from the Trinamic drivers, only if you want to support per motor fault signals from external drivers. I have not yet added support for per axis warning signals from external drivers.
Fault signal handling for the SLB EXT board is currently in [board specific code](https://github.com/grblHAL/STM32F4xx/blob/2af281736ab0cc51a9947fc80fdb41ed21af46cf/boards/longboard32.c#L118-L164) and the inputs are polled due to limited support for interrupt capable pins in STM32 MCUs. I'll likely split out this code to either _driver.c_ or to a new file at some point.

dresco, 2025-01-22:
Following on from the issue in the Trinamic repo..

I've set up a WeAct dev board with a single external 5160 driver ($338=1). The WeAct map is configured for both software and hardware SPI on the same pins, so can easily swap between implementations by defining `TRINAMIC_SOFT_SPI`. 

However, am still unable to replicate this issue either way. I have sigrok traces for both setups, and can also get a saleae logic trace if helpful. 

@Freezerohmatic would be helpful to better understand the timings that seem to break your configuration, this is what I see in my testing;

HW SPI: chip select is driven high for a minimum of ~120nS between each transaction, and the first SPI clock rising edge is ~4uS after CS is driven low. Data rate is ~1.5MHz.

SW SPI: CS has similar timing (minimum ~100nS), first SPI clock rising edge is ~450nS after CS is driven low. Data rate is ~1.2MHz.

Freezerohmatic, 2025-01-23:
So my current set up is using SPI 1 (GPIOA: SCK = 5, MISO = 6, MOSI = 7) and I've done most of my testing on the X Axis CS pin, GPIOC 4. When testing each axis seperatly none would initialise at boot but they would after spamming M122I a lot. Spamming M122I with multiple axis enabled would always make it fail but at different motors. I added a debug output to help me track this down. 

I added delay after each `DIGITAL_OUT(cs[driver.id].port, 1 << cs[driver.id].pin, x);` in tmc_spi.c and this made it 100% reliable. I did this as I did not see the X Axis CS pin go low using my oscilloscope. 

I'm not an ARM dev, I spend most of my time in PICs so I have no idea how the IO bus on the STM32H723 behaves.

I'm trying some of your suggestions from the Trinamic repo and reading up on how the STM32H723 IO bus works to see if I can gain a better understanding on whats going on.

Freezerohmatic, 2025-01-23:
Removed delays and changing to software SPI. 
I get the same results as dresco and all drives initialise on boot as expected.

Setting L1_CACHE_ENABLE = 0 no delay
CS pins do not change. SPI is transmitting

Setting L1_CACHE_ENABLE = 0 with delays
all drives initialise on boot as expected. Delay between CS low and SPI clock start is 7uS

HW SPI using __DSB();         
CS pins does not change. SPI is transmitting

HW SPI using a blocking while loop to check the pin state         
CS pins does not change. SPI is transmitting

That last one makes no sense so I'm guessing that port->IDR is reading the register state rather then reporting the current pin state similar to PORTX vs LATX in PICs

Freezerohmatic, 2025-01-23:
@dresco I notice that I had made mistake with my board config I set it up for generic_stm32h723vg instead of generic_stm32h723ze but I'm guessing thats just for flash size? It does not change the result of my last comment

Freezerohmatic, 2025-01-23:
changing SPI_BAUDRATEPRESCALER_32 to SPI_BAUDRATEPRESCALER_16 reduces CS low to SPI clk start to 2us

dresco, 2025-01-23:
> set it up for generic_stm32h723vg instead of generic_stm32h723ze but I'm guessing thats just for flash size?

Yep, only differences in the linker files are the flash size & EEPROM emulation.

The h723ze file does not have flash set aside for EEPROM emulation, as only 512KB available. Is not needed (so far) as both boards using this part already have a serial EEPROM onboard.

dresco, 2025-01-23:
@Freezerohmatic Just for clarity, when using HW SPI your CS pin stays high all the time - unless you add these additional delays after setting it low?

Am struggling to think of good reason for the behaviour you're seeing. There are some problems with your configuration - external interrupt clashes etc, but don't think that is anything that would explain this..

Can you build & send me a binary using the following build environment, and will see if behaviour differs to what I'm building locally? I notice you've overridden the `OVERRIDE_MY_MACHINE` common setting, so you'd need to add that back in first..

```
# WeAct MiniSTM32H723 development
[env:weact_mini_h723_dev]
board = generic_stm32h723vg
board_build.ldscript = STM32H723VGTX_FLASH.ld
build_flags = ${common.build_flags}
              ${usb_h723.build_flags}
              ${sdcard.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_WEACT_MINI
  -D HSE_VALUE=25000000
  -D RTC_ENABLE=0
  -D TRINAMIC_ENABLE=5160
  #-D TRINAMIC_SOFT_SPI=
  -D SPINDLE1_ENABLE=SPINDLE_PWM0
lib_deps = ${common.lib_deps}
           ${usb_h723.lib_deps}
           ${sdcard.lib_deps}
           motors
           trinamic
lib_extra_dirs = ${common.lib_extra_dirs}
                 ${usb_h723.lib_extra_dirs}
                 ${sdcard.lib_extra_dirs}
upload_protocol = stlink
debug_tool = stlink

```

Freezerohmatic, 2025-01-24:
That is correct however if I enable one trinamic at a time I sometimes see the pin go low. It honeslty feels random and I don't really have any enviromental variables to blame. 

See [link](https://github.com/Freezerohmatic/STM32H7xx_grblHAL/releases/tag/BOARD_WEACT_MINI_TMC) for binary files

I made 2 versions, one with delays and one with out for your refernce. 

Re the board config, I wouldn't mind fixing those issues, I'm guessing limit X and limit M3 are an issue for some reason but I'll jump to the other issue for that. 

For myself I'm going to just use the added delays, from what you are saying about your config without them, it doesn't add much overhead and I'm a little tired of trying to figure out whats going one, I have other issues that I would like for focus on, like why sensorless homing isn't playing nice but SG seems to work fine in ioSenders Trinamic tuner and the SD card not seeming to work. 

I'll clean up the added delays and make it a setting to enable. Maybe we could get @raoullector to try my work around to see if this resolve their issue rather then having to enable it one by one.

dresco, 2025-01-24:
Will check the attached files a bit later, thanks..

> I'm guessing limit X and limit M3 are an issue

It's reset and M3 limit that are clashing. On STM32, external interrupts are tied to pin numbers, regardless of port. So can't use (for example) both PG11 & PE11.

I've started adding the necessary bits to my local repo, will go through your map file in more detail and see what may need changing. Unfortunately there is not much input protection on this board (as with many of these 3d printer based boards). There is just one opto-isolated input by the looks of things, but there are a few MCU pins wired directly to the expansion headers. I believe you are first user of this board, so happy for the board map to be led by your wiring requirements.

> SD card not seeming to work.

SD should work. Only 4bit SDMMC access is currently supported in the H7 driver, but your board is correctly wired for this. From your map file, it looked like you were maybe trying to configure for SPI mode?

raoullector, 2025-01-24:
Hi Freezerohmatic,

> Maybe we could get @raoullector to try my work

Yes, I would like to perform this test, but my board is an SKR3 and not a WeAct. Even though the processor is the same (STM32H723), I fear that the test will mostly highlight my limitations in porting a given environment to another.

dresco, 2025-01-24:
@Freezerohmatic Probably worth noting that the SKR3 is software SPI only..

dresco, 2025-01-24:
Have made some progress troubleshooting this afternoon..

My setup for testing is an external 5160 breakout with VM powered from a 12V bench supply, and 3.3V VIO powered from the MCU board. Trinamic driver is configured to use hardware SPI.

[image]

Observations are as follows;

If VM on the 5160 **is** powered before the MCU board is powered, then is all good. Initialisation always works.

If VM **is not** powered before the MCU board is powered, then as expected the initial initialisation fails. However, once power has been applied to VM, subsequent attempts to initialise will also often fail, until the MCU board is reset.

In my testing, this is the situation where the additional delay after toggling  the CS line makes a difference. With the additional delays in place, this subsequent initialisation always works. (Still not sure why, but at least there is an explanation for the difference in behaviour).

Other things I tried that **didn't** help included a soft reset from grblHAL (Ctrl-C), and disconnecting/reconnecting the connections between the MCU board and 5160 breakout.

Note that this only applies to hardware SPI, I've still not been able to recreate this issue at all using software SPI (as used on the SKR3).

Freezerohmatic, 2025-01-25:
`Probably worth noting that the SKR3 is software SPI only..` Ah, didn't reliase that.. though at this rate who knows.

With my setup I'm powering the board via the 12V input which has buck converters for 5v and 3v3. Vcc seems to raise to 5v at 8.5v on VIN. Which should be high enough for the 5160.

dresco, 2025-01-25:
Found the problem I think..

The CS high time is too short between subsequent calls to tmc_spi_read() & tmc_spi_write() functions.

It is set high immediately before returning from both these functions, which results in some of the high times being only ~100nS, instead of the expected ~400nS with the corresponding delay.

The strange thing is that this only causes a problem sometimes! Possibly related to the timing of the clock signal & this description in the datasheet (although I couldn't reliably demonstrate this);
[image]

Regardless, can see that we're not receiving the expected status byte on the transaction following the short pulse, presumably because the driver has not recognised the end of the previous one..

I'll do some more testing of all the different scenarios, but the following change seems to reliably fix this for me;

```
diff --git a/Src/tmc_spi.c b/Src/tmc_spi.c
index e5e22cd..b6e8914 100644
--- a/Src/tmc_spi.c
+++ b/Src/tmc_spi.c
@@ -139,6 +139,7 @@ TMC_spi_status_t tmc_spi_read (trinamic_motor_t driver, TMC_spi_datagram_t *data
     datagram->payload.data[0] = spi_get_byte();
 
     DIGITAL_OUT(cs[driver.id].port, 1 << cs[driver.id].pin, 1);
+    delay();
 
     return status;
 }
@@ -157,6 +158,7 @@ TMC_spi_status_t tmc_spi_write (trinamic_motor_t driver, TMC_spi_datagram_t *dat
     spi_put_byte(datagram->payload.data[0]);
 
     DIGITAL_OUT(cs[driver.id].port, 1 << cs[driver.id].pin, 1);
+    delay();
 
     return status;
 }
```

dresco, 2025-01-26:
@raoullector @Freezerohmatic 

I've updated my H7 repo with the latest CS timing fix for 5160 (have stretched it further, to bring into line with the datasheet typical timing). Tested successfully here on WeAct dev board, BTT Scylla, and BTT SKR3.

Would be grateful if you would also be able to test..

raoullector, 2025-01-27:
My tests are OK, thank you Jon.

On my setup (SKR3, STM32H723, 5x EZ5160 with one ganged axis), the drivers are automatically initialized.
The initialization delay (measured by ear) between powering up the machine and the "clack" of torque engagement is less than one second.

Perfect, as far as I’m concerned this ticket can be closed.

Freezerohmatic, 2025-01-28:
I can confirm that this works on my board.

dresco, 2025-01-28:
Thanks both. I don't have permissions to close the ticket, so feel free..