grblHAL/Plugins_motor #20: TMC2209 needs to be reinitialized after sensorless homing

troth530, 2025-01-27:
Sensorless homing works for a single axis but afterward requires a M122i command before hard limits will work for stall detection when it hits a hard stop.   Also, homing X and Y axis for instance in the same cycle ($44=3) will result in sensorless homing not working.  What happens is diag pin will trigger on the first axis which hits a hard stop during the seek operation,  but diag pin will not trigger on the second axis and the homing operation will not complete.   I believe these are part of the same issue.  For now my workaround is set $44=1 and $45=2 and after a homing cycle type M122i.  In this case homing completes and hard limits are triggered when it hits a hard stop.  This is of course with the correct limit settings as well as the correct thresholds that work for single axis homing.

I am testing on ESP32 S3 on a single pin UART for all 4 drivers.

Thank you!

terjeio, 2025-02-04:
I just have commited an update that _may_ work better, more info [here](https://github.com/grblHAL/Plugins_motor/issues/21#issuecomment-2634822309).

I still do not have a board to test StallGuard homing with TMC2209 drivers, they are special in that the diag output is not open drain like the other drivers I have so are not possible to wire-or with regular limit switches.

troth530, 2025-02-10:
Hi @terjeio, I am unable to test for now as I got stuck on an issue with your latest changes.... seems USB_SERIAL_CDC is now crippled.  During linking I get an error, odd that it reports the error a few lines before the function call.  Setting USB_SERIAL_CDC to 0 is able to compile, I only get this error when using the S3 native USB.  

[image]

Can you please look into this.  Thank you!

troth530, 2025-02-10:
> It is a regression from the [last PR](https://github.com/grblHAL/ESP32/pull/137), move [this line](https://github.com/grblHAL/ESP32/blob/cc2a62be9075646f23526ce40dabd9ec9e04df21/main/usb_serial.c#L46) to line 33 to fix.

Thanks, that fixed USB_SERIAL_CDC issue.  I will test and report back later this week.